UBCSailbot/sailbot_workspace #37: Make catkin_ws/src a Docker volume

patrick-5546, 2022-10-04:
### Purpose
Make the ROS 1 workspace persistent so that uncommitted changes survive rebuilds.

### Resources
- https://code.visualstudio.com/remote/advancedcontainers/add-local-file-mount
- Volumes that are currently set up in the dev image and devcontainer.json: bash history and ros logs

patrick-5546, 2022-10-23:
Test if this works across dev images

DFriend01, 2022-10-30:
@Harvir14 If we clone our repositories in the [dev container image](https://github.com/UBCSailbot/sailbot_workspace/blob/main/.devcontainer/dev/Dockerfile), we could potentially lose work when updating the image and rebuilding the container.

Rather than cloning during the build process, we should just write a script that checks if the directory is cloned, and if it is not, then the directory is cloned. Otherwise, we do not clone it.

Write a script for the ROS1 workspace (`/workspaces/catkin_ws/src/`) and for the ROS2 workspace (`/workspaces/sailbot_workspace/src/`). Then, we would want to remove the lines below and build the image again:

https://github.com/UBCSailbot/sailbot_workspace/blob/4b35bea3e1b8de4e2c03f25dadaece7748791947/.devcontainer/dev/Dockerfile#L105-L107

@patrick-5546 Does this sound about right?

patrick-5546, 2022-10-30:
Yes, but this would only be necessary if the current method doesn't work. See if the current method persists across dev images first.

Harvir14, 2022-11-05:
Devon and I confirmed that the issue does not exist when you pull and build a different Dev Image from Github, so we can close this issue.

PR #44 (2022-10-22): Create Docker volume for ROS 1 workspace
This branch resolves #37 .
closed 2022-11-05 by Harvir14