UBCSailbot/sailbot_workspace #64: Create "wipe" VS Code task

patrick-5546, 2022-11-26:
### Purpose

When switching to a branch with different build dependencies, it may be necessary to wipe the workspace of all generated code.

### Changes

- Run "clean" task
- Run "purge" task
- Delete repositories in `src/` **only if there are no uncommitted changes, else fail**

### Relevant files
- Tasks are in the VS Code workspace file

DFriend01, 2024-01-01:
@patrick-5546 Is it fine to just combine clean and purge into one task, or do you still want to wipe out directories in `src`? I'm just tired of running both clean and purge, so I want a task that does both :-)

patrick-5546, 2024-01-01:
Yea feel free to assign yourself and make a PR

DFriend01, 2024-01-01:
Turns out that `purge` is a more thorough version of `clean`. If we are not implementing the third change specified in the issue documentation, then there is no point in implementing `wipe`.

patrick-5546, 2024-01-01:
Yea I don't see a usecase for the third point. If you don't either, you can close this PR and use purge instead of clean+purge

DFriend01, 2024-01-01:
I briefly tried a solution but it's overcomplicated for what it's worth. I'm opting to close this issue.

closed 2024-01-01 by DFriend01