UBCSailbot/sailbot_workspace #78: Add back ament_pep257

patrick-5546, 2023-01-19:
### Purpose

The ament_pep257 linter was removed from our CI in #77 because it is broken. Once the fix has been published, we should add it back.

### Changes

- `linter: [lint_cmake, xmllint, flake8, pep257]` in tests.yaml
- Make ament_pep257 required in branch protection rules

### Resources

- The latest version of ament_lint for humble is 0.12.4: https://index.ros.org/p/ament_lint/#humble-overview
- The fix can be found in 0.12.5: https://github.com/ament/ament_lint/compare/0.12.4...0.12.5

hhenry01, 2023-01-28:
Ditto for the network-systems repo.

patrick-5546, 2023-01-28:
Is there a way to install the latest version if it's not in apt repostiory?

PR #98 (2023-02-26): Add ament_pep257 to Tests workflow

- Resolves #78
closed 2023-02-26 by patrick-5546