UBCSailbot/sailbot_workspace #80: Automate building and running base image on RPi with Docker CLI

hhenry01, 2023-01-28:
### Purpose

Our RPi needs to operate without a GUI, and just needs to run our software. Dedicated scripting to build and run will make the process a lot faster.

### Changes

New script(s) to build the container, run it, and also execute a list of programs when given the paths/commands (a default to run all software should also be defined).

### Resources

- https://docs.docker.com/engine/reference/commandline/cli/

patrick-5546, 2023-02-11:
Another solution to evaluate against building on RPi is to build in CI:

- Add workflow to push built image: clones repos and builds them
    - From base image
    - Runs on push to main and pull request
    - Tag is commit hash
        - Can also use `latest` tag to quickly test changes that you make

Upside is that you don't have to build or develop on RPi. Downside is that it may take a couple minutes for the images to be built and pushed.

If we build on the RPi, we can clone sailbot_workspace locally and make it a volume when running the base image, effectively making our own dev container

patrick-5546, 2023-02-11:
- Setup Docker on Rpi
- See how long it takes to build compared to x86 laptop
- Decide on path forwards: build locally, CI, or on x86 laptop?

patrick-5546, 2023-02-13:
### Build Times

| Laptop Model | Build Time | CPU | Memory | Architecture | OS | Other Notes |
| - | - | - | - | - | - | - |
| XPS 15 | 30 s | i7-12700H | 32 GB | x86 | Windows | |
| Lenovo C940 | 50 s | i7-1065G7 | 16 GB | x86 | Windows | |
| Lenovo C940 | 50 s | i7-1065G7 | 16 GB | x86 | Ubuntu | Used Docker Desktop which runs in a VM, Docker Engine may be faster |
| Macbook Pro 14 | 30 s | M1 Pro | 16 GB | arm | MacOS |
| Macbook Air | 2 min | M1 | 8 GB | arm | MacOS | Memory was bottleneck: even before running docker memory was full, peaked at 11GB swap |

patrick-5546, 2023-02-13:
@hhenry01 do you have docker engine or desktop installed on popos? Can you add your build times?

hhenry01, 2023-02-13:
> @hhenry01 do you have docker engine or desktop installed on popos? Can you add your build times?

Docker engine. Added.

PR #94 (2023-02-26): Add script to run container via docker cli

### Description

- Resolves #80 
- Created deployment_scripts/ directory as I anticipate there will be more deployment specific scripting in the future, like running a set of programs
- deployment_scripts/start_container.sh can either build and run the latest version of the workspace that is cloned, or it can run an arbitrary version built in the past
- Docker run arguments defined under the "runArgs" field in devcontainer.json are automatically parsed and used when starting the container, but there is a caveat with respect to certain devcontainer.js [... cut]
closed 2023-02-26 by hhenry01