UBCSailbot/sailbot_workspace #101: Add docs site

patrick-5546, 2023-03-01:
### Purpose

Able to develop docs site inside sailbot_workspace.

### Changes

- [ ] Clone docs site
    - Easiest would be for this to be defined in [`new_project.repos`](https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/new_project.repos), but I'm not sure if the workspace will still build with non-ROS packages in the `src/`
    - If it can't build when in `src/`, clone in `~/`
- [ ] Add path to docs repository to `folders` in [`sailbot_workspace.code-workspace`](https://github.com/UBCSailbot/sailbot_workspace/blob/main/.devcontainer/config/sailbot_workspace.code-workspace)
- [ ] Add VS Code config in [`.vscode/` of docs](https://github.com/UBCSailbot/docs/tree/main/.vscode) except extensions to [`sailbot_workspace.code-workspace`](https://github.com/UBCSailbot/sailbot_workspace/blob/main/.devcontainer/config/sailbot_workspace.code-workspace)
    - Extensions would go in [`devcontainer.json`](https://github.com/UBCSailbot/sailbot_workspace/blob/main/.devcontainer/devcontainer.json)
- [ ] `pip install -Ur <path to docs requirements.txt>` in [`setup.sh`](https://github.com/UBCSailbot/sailbot_workspace/blob/main/setup.sh) inside VCS if statement after VCS command
- [ ] Add [apt dependencies](https://github.com/UBCSailbot/docs/blob/6b5d0aeb8ab24e8ecd8be5bca4038a787efd6e0b/.github/workflows/publish.yml#L30) to [dev image dockerfile](https://github.com/UBCSailbot/sailbot_workspace/blob/main/.devcontainer/base-dev/base-dev.Dockerfile)

patrick-5546, 2023-04-08:
Dependencies should go in `local-base` image

DFriend01, 2023-05-15:
The PR for this issue can be merged without completing the item on the last list. There is an existing issue for it [here](https://github.com/UBCSailbot/docs/issues/85)

PR #115 (2023-05-22): Add docs site

### Description

- Resolves #101

Refer to the checklist in #101. 

I wasn't able to programmatically add the vscode settings files from the docs repo to the codespace JSON file, so I'll probably need some help there. The not so great alternative is to copy and paste.
closed 2023-05-22 by DFriend01