UBCSailbot/sailbot_workspace #350: Mast Wind Sensor Angle Correction

hhenry01, 2024-03-30:
### Purpose

### Description

Since the mast wind sensor will rotate with the mast, the angle it outputs will not be constant with respect to the heading of Polaris. This should be pretty simple, just take the angle of the mast wind sensor and the sail encoder and do some simple trig such that an output wind angle fits the [message description](https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/custom_interfaces/msg/WindSensor.msg). *Note*: the sail encoder angle conventions need to be defined with ELEC.

### Resources

-

shihlings, 2024-06-24:
Unsure if SOFT should be dealing with this. I believe that ELEC is already doing the sail angle compensation for the mast wind sensor.

FireBoyAJ24, 2025-07-12:
@MichaelGreenough

FireBoyAJ24, 2025-07-12:
Handled by ELEC team.

closed 2025-07-12 by FireBoyAJ24