UBCSailbot/sailbot_workspace #363: Unit tests start 2 http servers on the same port

patrick-5546, 2024-04-25:
### Description

A recent NET PR causes the test task to fail: both NET and PATH unit tests start http servers on port 8081

#### Expected Behavior

Test task should pass

#### Actual Behavior

On my machine the local pathfinding unit tests run after net, and I get this error:
[image]

### Steps to Reproduce

1. Run the test task

### Relevant Code

- NET: https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/network_systems/scripts/http_echo_server.py
- PATH: https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/local_pathfinding/test/post_server.py
    - Temporary substitute for NET global path endpoint

### Potential Solutions

@hhenry01 any ideas? Is the NET global path endpoint working yet? Or should one of the servers run on a different port

hhenry01, 2024-04-25:
I'm assuming this happens because both the NET and PATH unit tests are ran in parallel. We would either need to run the servers on different ports for unit testing only or run the NET and PATH tests sequentially.

patrick-5546, 2024-04-25:
What's the net server for? Is it supposed to mimic the webhook server as well which will run on 8081?

hhenry01, 2024-04-25:
Yes. Virtual Iridium needs an active server or else it will crash.

patrick-5546, 2024-04-25:
@SPDonaghy we should use a different port then until it's implemented on the net side

SPDonaghy, 2024-04-25:
> @SPDonaghy we should use a different port then until it's implemented on the net side

Sounds good, I can make that change tomorrow

PR #365 (2024-04-29): Changed Port for Posting Global Paths in Tests

### Description

- Resolves #363 

Moved the test post_server from port 8081 to 8085.
Also refactored the test files a bit so that we only need to change one line to change the test port again if needed.

### Verification

- Ran the test task, but the local path tests might be running before NET on my machine
closed 2024-04-29 by SPDonaghy