UBCSailbot/sailbot_workspace #425: Update low level control node with implemented controllers

eomielan, 2024-08-06:
### Purpose
The purpose of this issue is to update the low level control node (`boat_simulator/nodes/low_level_control/low_level_control_node.py`) to use the implemented rudder and sail actuation controllers.

### Changes
Update `__rudder_actuation_routine()` to use the implemented rudder controller and `__sail_actuation_routine()` to use the implemented sail controller.

### Resources

- [Controllers Implementation](https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/boat_simulator/boat_simulator/nodes/low_level_control/controller.py)
- [Low Level Control Node](https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/boat_simulator/boat_simulator/nodes/low_level_control/low_level_control_node.py)

alberto-escobar, 2024-09-17:
# low_level_control_node update notes

add `From controller import RudderController, SailController`

create initialization of RudderController and SailController, only one instance of each within the node

__rudder_actuation_routine:
- parameters obtained from ros
- current and desired found similar to below
		
	

__sail_actuation _routine:
- Scontroller.reset_setpoint if new value
- Scontroller.update_sate() ???how to loop
- - loop 
	Scontroller.calculate error
	??? Where do I get the target_angle and current_control_ang
		self.sail_trim_tab_angle(self)
		from goal_handle
			??? how to acces parameter .request

questions for eric:
how to access desired heading/angle from goal_handle?
	-.request
where to get kp, cp, speed, other stuff?
	ROS parameters, access form. chekc physics engine node line 152

should I loop update_state() in the loop in the routine or make a new loop?
	new loop
should I be removing the SimSailTrimTabActuation calls
	keep it

alberto-escobar, 2024-09-21:
Testing blocked by errors in physics engine node

closed 2025-11-23 by alberto-escobar