UBCSailbot/sailbot_workspace #497: Make CAN Transceiver run on loopback CAN

samdai01, 2025-03-01:
### Purpose
As the first step of ELEC integration, we need to ensure that the CAN transceiver can properly start when we do `ros launch network_systems main_launch.py`. This means going to the EDC, running sailbot workspace on the raspberry pi, starting CAN in loopback mode and ensuring that the CAN transceiver can bind to the can socket. 

The end result should be:
- run `ros launch network_systems main_launch.py`
- no errors with binding to can socket
- run CAN in loopback mode
- use `cansend` to send a test CAN frame
- verify that can frame is logged in ROS through publishing from CAN Transceiver

### Resources
https://www.pragmaticlinux.com/2021/10/can-communication-on-the-raspberry-pi-with-socketcan/

adambrett40, 2025-03-23:
Battery, GPS, Wind, AIS, GenericSensor are what we can test from cansend, because they receive from can bus

Need to add error handling so can_transceiver doesn't stop when receives bad frame

review logging messages (wind is currently printing m/s instead of km/h

Battery:
- doesnt work rn cuz it sends data to the can bus, which hangs waiting for a response

GPS
- works, speed placeholder bounds are bad

Wind
- works

AIS
- not yet tested

GenericSensor

need to add ros callbacks for new sensors, see if we can do START, END ranges instead of 16 individual
ask about new ros topics/publishers

lross03, 2025-03-23:
Also need to make ROS callback for rudder data frame

lross03, 2025-03-24:
> Battery, GPS, Wind, AIS, GenericSensor are what we can test from cansend, because they receive from can bus
>
> Need to add error handling so can_transceiver doesn't stop when receives bad frame [...]

Note we can also test rudder_data with cansend, but it currently doesnt have a ros callback as mentioned

adambrett40, 2025-03-29:
TODO:
- investigate rpi slow/failing. overheating? memory capacity?
- run script for testing receiving frames
- make script for testing sending frames
- get new frames approved
- get error handling finished/approved
- fix placeholder bounds with feedback from elec
-