UBCSailbot/sailbot_workspace #531: Advanced mock wind sensor and gps nodes

SPDonaghy, 2025-03-24:
### Purpose

Upgrade the mock wind sensor and gps nodes to publish more realistic data that depends more on previous values.

### Changes

### Mock WindSensor
**Goal:** we want to be able to pass in arbitrary mean wind speed and mean direction values from our test plans.

**Idea:** Configure the mock wind sensor node's mean wind speed and mean direction to be read from a csv file whose file path is determined from a ROS parameter so that we can change it during runtime and switch to different wind conditions without editing any code.

Other ideas?

### Mock GPS
Assuming sailbot perfectly achieves the previous desired heading as its new current heading is good enough.

**Goal:** We want the boat speed to update dynamically depending on mocked wind conditions.

There is a table of predetermined values for expected boat speed depending on wind speed and direction. You can find this table [HERE](https://github.com/UBCSailbot/sailbot_workspace/blob/1f2c97b3c528366b8ac7cb7899b3ac557db5c43c/src/local_pathfinding/local_pathfinding/objectives.py#L21-L32)

**Idea:** We can calculate the mock gps boat speed by using the random mock windsensor data and an interpolation of the boat speed data provided above.

raghumanimehta, 2025-03-24:
Sure, I'll get started on this soon.

SPDonaghy, 2025-03-26:
Just had a thought for this, could you make it so that whenever the ros parameter is set, the node just publishes that parameter value constantly until the parameter is set back to some default out of range value?

raghumanimehta, 2025-04-19:
Sure! I can do look into that, shouldn't be hard. I have a concern about visibility though, but I'll look into it.

raghumanimehta, 2025-04-20:
> **Idea:** Configure the mock wind sensor node's mean wind speed and mean direction to be read from a csv file whose file path is determined from a ROS parameter so that we can change it during runtime and switch to different wind conditions without editing any code.
> 
> Other ideas?
About this, do you want the CSV to contain the mean speed value and direction for cold start or do you want to be able to read all the values from CSV eg. instead of calculating the node just reads the values and publishes it? 
If you want to do the first option, don't you think it would be better if we can just use `ros param set \mock_wind_sensor_node speed 5` to change the mean speed to 5?
This allows us to update the mean speed (and direction) on the fly without having to read from csv. Let me know what you think about this.

SPDonaghy, 2025-04-21:
> we can just use ros param set \mock_wind_sensor_node speed 5 to change the mean speed to 5

Yeah I think that is a great approach, as long as there is a way to switch between letting the direction/speed vary vs. fixing them at constant values.

PR #553 (2025-04-28): Raghumanimehta/531 upgraded mock gps wind sensor

### Description

- Resolves #531
- Mock wind sensor data can be set during runtime now by using `ros2 param set /mock_wind_sensor mean_wind_speed {value}` and  `ros2 param set /mock_wind_sensor mean_direction {value}`.
- Fixed a bug with `ensure_correct_range` in `node_mock_wind_sensor.py` that was not handling direction values less than -179.
closed 2025-04-28 by SPDonaghy