UBCSailbot/sailbot_workspace #532: Optimize objective cost weights for path planner

SPDonaghy, 2025-03-24:
### Purpose

Our path planner attempts to find the optimal path based on our custom optimization objectives which are defined in `objectives.py`. The planner calculates the total cost for any path as a single float value which is equal to the weighted sum of costs, along the entire path, for each objective.

Right now the weights for those path objective costs are just best guesses. 

We want to optimize these weight values to be sure that we are building the best possible (or close to it) pathfinding model that we can.

### Changes

Change the weights of the objective costs to our new optimized values.

### Resources

- [objectives.py](https://github.com/UBCSailbot/sailbot_workspace/blob/main/src/local_pathfinding/local_pathfinding/objectives.py)
- [OMPL Optimal Planning](https://ompl.kavrakilab.org/optimalPlanning.html)
- [Hyperparamter Optimization](https://en.wikipedia.org/wiki/Hyperparameter_optimization)

jamenkaye, 2025-04-30:
Discussion of this issue on [Confluence](https://ubcsailbot.atlassian.net/wiki/spaces/prjt22/pages/2503704581/Path+Cost+Objective+Weights?atlOrigin=eyJpIjoiYzZlNzRjYmVkMmNlNDM5NzhmZjE5MGMyMTljYzU3MTUiLCJwIjoiYyJ9)

FireBoyAJ24, 2025-07-12:
Linked back to this issue: [#290](https://github.com/UBCSailbot/sailbot_workspace/issues/290)

raghumanimehta, 2025-12-25:
Closing this because we are doing a major rework of objectives. Will redefine this problem later as per the updated requirements.

closed 2025-12-25 by raghumanimehta