UBCSailbot/sailbot_workspace #565: Integrate Sailbot Simulation into Pathfinding Visualizer

FireBoyAJ24, 2025-05-23:
### Purpose

Integrating the sailbot's simulator allows us to model the boat's movement based on the changing wind speeds and conditions rather than assuming that the boat will perfectly sail from one waypoint to the next.

### Description

The observer node can also directly subscribe to the GPS topic when the local pathfinding mock GPS is deactivated. The simulator and network system nodes will run and publish the sailboat's filtered wind sensor and GPS at regular intervals. These two nodes will act as the boat's physical system from which the pathfinding system has to find the local waypoints. 

### Resources

- [Boat Simulator Design Specification](https://ubcsailbot.atlassian.net/wiki/spaces/prjt22/pages/1785790595/Boat+Simulator+Design+Specification)
- [Visualizer Design Specification](https://ubcsailbot.atlassian.net/wiki/spaces/prjt22/pages/2338455564/Visualizer+Design+Specification)

eomielan, 2025-05-28:
> [@eomielan](https://github.com/eomielan) does the simulator mock gps respond to the desired heading that pathfinding will publish?

Yes it should

SPDonaghy, 2025-12-25:
Yes, the decision was made for pathfinding to implement their own mocks as a means to move forward with testing as the Simulator project was not in a complete enough state for the pathfinding team to use it.

closed 2025-12-25 by SPDonaghy