UBCSailbot/sailbot_workspace #605: Verify/modify filtered wind sensor

lross03, 2025-07-19:
### Description

Pathfinding wants only filtered wind sensor to be published - meaning filter out erroneous data and only give the average. We need to filter out bad data   and only send valid. So in the event of 1/2 sensors sends bad data, send just the other sensor.

adambrett40, 2025-10-17:
#271 mentions the idea of using a moving average of sensor data to help reduce noise, which would be implemented in an abstracted library. 
Our current implementation (placeholder) will attempt to construct a wind frame when received, then publish the average of the two most recent wind frames (SAIL_WIND and DATA_WIND respectively).

e.g.
receive SAIL_WIND: 10kts -> publish (10 + 0)/2
receive DATA_WIND: 12kts -> publish (10 + 12)/2
receive DATA_WIND: 13kts -> publish (10 + 13)/2
and so on.
Problem is if one sensor breaks or starts sending garbage data, in which case it's most recent will never update, and a 'ghost' value will continue to contribute to the filtered wind reading.

For now I'm just implementing a simple moving average hardcoded for wind_sensor, and if we end up needing it for another sensor then we can abstract to a library.

What we need to know in order to figure out how many data points we want/what kind of filter would be best:
- how often is wind sensor data being published
- how noisy is the individual sensor data
- how much variation is there between the two different sensor readings

These would be good things to figure out from on-water testing, unless ELEC already has some readings from the first test.

Also relevant: currently wind_sensor.direction is int16, which means when taking any kind of average of multiple sensors will lose precision. I'm gonna change it to float32 since I don't think there's any reason it needs to be int16 other than that we didnt think more precision was necessary.

adambrett40, 2025-10-17:
Just realized we actually want a moving _vector_ average of course, since the speed is correlated with the direction, and taking flat averages of angles is silly.